koide3/glim #238: glim node got killed

vickzk, 2025-07-02:
```
[glim_rosnode-1] terminate called after throwing an instance of 'gtsam::IndeterminantLinearSystemException'
[glim_rosnode-1]   what():  
[glim_rosnode-1] Indeterminant linear system detected while working near variable
[glim_rosnode-1] 8502796096475496449 (Symbol: v1).
[glim_rosnode-1] 
[glim_rosnode-1] Thrown when a linear system is ill-posed.  The most common cause for this
[glim_rosnode-1] error is having underconstrained variables.  Mathematically, the system is
[glim_rosnode-1] underdetermined.  See the GTSAM Doxygen documentation at
[glim_rosnode-1] http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
[glim_rosnode-1] more information.
```
Facing the above error and node got killed randomly

soumya997, 2025-07-11:
Okay, figured out,
I think the T_lidar_imu value need to be changed to this,

```
"T_lidar_imu": [
      0.006,
      -0.012,
      0.031,
      0.0,
      0.0,
      1.0,
      0.0
    ]
```

follow this: https://github.com/koide3/glim/issues/197

vigneshmgz, 2025-07-11:
It should throw appropriate warning/error instead of getting killed

liuxin-SH, 2025-09-13:
I'm also facing the issue